RoboticsBrno/Robutek #21: Add name and version label

dzarda, 2024-05-22:


TVavrinec, 2024-05-23:
The question is, what do we want to write next?
[image]

dzarda, 2024-05-23:
Designed by Tomáš Vavrinec

TVavrinec, 2024-05-24:
What do you think about that?
[image]

dzarda, 2024-05-27:
Maybe add "IOxx" labels to sensors

dzarda, 2024-05-27:
Add servo and iled labels from top side.

1. I want to know what the connector is (ILED, SERVO)
2. I want to know the pin it's connected to

TVavrinec, 2024-05-29:
Servo labels
[image]

Sensor labels
[image]

closed 2024-06-07 by TVavrinec