RoboticsBrno/Robutek #30: Increase BAT_SENSE divider reduction

dzarda, 2024-05-25:
[image]
This gets you 3.318V for 4.35V cell voltage, which is bad.

TVavrinec, 2024-05-27:
I think you have the wrong version
[image]

TVavrinec, 2024-05-27:
What do you think about that?
[image]

dzarda, 2024-05-27:
That would not work. I'd go with 2x 17.4k

TVavrinec, 2024-05-28:
Oh, I switched them, sorry I missed the text under the picture.
If I want 0.95V at the output (so I don't need a divider in the ESP), I want this.
[image]

dzarda, 2024-05-29:
ok

dzarda, 2024-05-29:
Actually this gives you tau of cca 1s. Isnt it too much?

TVavrinec, 2024-05-31:
[image]

closed 2024-05-31 by TVavrinec