RoboticsBrno/Robutek #31: What's the deal here?

dzarda, 2024-05-25:
[image]

dzarda, 2024-05-25:
[image]

dzarda, 2024-05-25:
[image]

dzarda, 2024-05-25:
[image]

dzarda, 2024-05-26:
[image]

dzarda, 2024-05-26:
[image]

dzarda, 2024-05-27:
[image]

dzarda, 2024-05-27:
[image]

dzarda, 2024-05-27:
[image]

TVavrinec, 2024-05-27:
https://github.com/RoboticsBrno/Zelvicka/issues/31#issue-2317285519
https://github.com/RoboticsBrno/Zelvicka/issues/31#issuecomment-2133852155
I don't know exactly what you mean

closed 2024-05-31 by TVavrinec